acfr/snark #93: gobi-cat: --set-and-exit doesn't work

junderwood, 2015-01-21:
using --set-and-exit does not appear to cause the settings to be retained by the camera
using --set does work.

e.g. on shrimp (or ladybird with different IP)
##### the following does not work:
gobi-cat --address=192.168.3.3 --set-and-exit "IntegrationTime=20;MinimumFrameTime=100000"
#no need to ctrl-c because set-and-exit does exit as it should
gobi-cat --address=192.168.3.3 --list-attributes | egrep -i "IntegrationTime|MinimumFrameTime|ExposureTimeAbs

##### the following does work:
gobi-cat --address=192.168.3.3 --set "IntegrationTime=20;MinimumFrameTime=100000" > /dev/null
# wait for about 10 seconds and then hit ctrl-c
gobi-cat --address=192.168.3.3 --list-attributes | egrep -i "IntegrationTime|MinimumFrameTime|ExposureTimeAbs

andsok75, 2015-01-21:
Can you try
gobi-cat --address=192.168.3.3 --set-and-exit --set "IntegrationTime=20;MinimumFrameTime=100000"

I think --set-and-exit is just a switch that tells gobi-cat not to run the pipeline.

junderwood, 2015-01-21:
That worked, thanks!

closed 2015-01-21 by junderwood