acfr/snark #94: cmake: spherical_geometry causes cmake generate error

andsok75, 2015-02-25:
In ccmake ., running [c] configure and then [g] generate results in the errors shown below.
cmake version 2.8.2 (mantis.littleboard). The same errors are observed even if everything is disabled expect for BUILD_APPLICATIONS.

 CMake Error in math/CMakeLists.txt:
   Cannot find source file "$<TARGET_OBJECTS:snark_math_spherical_geometry>".
   Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
   .hxx .in .txx

 CMake Error in math/spherical_geometry/CMakeLists.txt:
   Cannot find source file "OBJECT".  Tried extensions .c .C .c++ .cc .cpp
   .cxx .m .M .mm .h .hh .h++ .hm .hpp .hxx .in .txx

vlaskine, 2015-02-25:
@andsok75, could you please talk to @navidp and fix this problem together? it should be trivial; thank you

andsok75, 2015-02-25:
The error I posted previously didn't comeout quite right. Here's the correct version
 CMake Error in math/CMakeLists.txt:
   Cannot find source file "$<TARGET_OBJECTS:snark_math_spherical_geometry>".
   Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
   .hxx .in .txx

 CMake Error in math/spherical_geometry/CMakeLists.txt:
   Cannot find source file "OBJECT".  Tried extensions .c .C .c++ .cc .cpp
   .cxx .m .M .mm .h .hh .h++ .hm .hpp .hxx .in .txx

andsok75, 2015-02-25:
github's markdown mangled one of the lines in the error message I posted. The line that reads 
Cannot find source file "$".
should have been 
Cannot find source file "$TARGET_OBJECTS:snark_math_spherical_geometry".

where I removed the brackets to make the error visible.

closed 2015-02-25 by andsok75