acfr/snark #95: spherical_geometry\coordinates.h: windows compile error

junderwood, 2015-03-11:
5>c:\users\j.underwood\code\cpp\source\snark\math\spherical_geometry\coordinates.h(57): error C2226: syntax error : unexpected type 'Eigen::Vector3d'
5>c:\users\j.underwood\code\cpp\source\snark\math\spherical_geometry\coordinates.h(57): error C2238: unexpected token(s) preceding ';'
5>c:\users\j.underwood\code\cpp\source\snark\math\spherical_geometry\coordinates.h(72): error C2226: syntax error : unexpected type 'snark::spherical::coordinates'
5>c:\users\j.underwood\code\cpp\source\snark\math\spherical_geometry\coordinates.h(72): error C2238: unexpected token(s) preceding ';'

vlaskine, 2015-03-11:
@navidp : could you please fix it? the problem is that on windows (sigh) "near" is a reserved word (i think)

the quick fix would be: rename the methods from "near()" to "is_near()" in snark::spherical::coordinates; and then fix compilation in aero

thank you!

navidp, 2015-03-11:
I renamed it to is_near. let me know if it solves the problem on windows.

closed 2015-03-11 by navidp

junderwood, 2015-03-11:
Thanks, that solved it.