acfr/snark #96: audio-sample: windows build fail

junderwood, 2015-03-11:
ClCompile:
  audio-sample.cpp
C:\Users\j.underwood\Code\cpp\libraries\boost\boost_1_55_0\boost/thread/pthread/mutex.hpp(10): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory

Isn't there a cross platform mutex from boost?

vlaskine, 2015-03-11:
it was a stray include, probably from some unfortunate autocompletion in the editor

closed 2015-03-11 by vlaskine

junderwood, 2015-03-11:
It explicitly only includes select on non-windows, for known reasons... so now the build fails on calls to select, which is undefined on win. Is it meant to compile on win or should it be removed from cmake if win32?

..\..\..\..\..\source\snark\audio\applications\audio-sample.cpp(182): error C2228: left of '.wait' must have class/struct/union
..\..\..\..\..\source\snark\audio\applications\audio-sample.cpp(183): error C2228: left of '.read' must have class/struct/union
..\..\..\..\..\source\snark\audio\applications\audio-sample.cpp(183): error C2228: left of '.ready' must have class/struct/union

vlaskine, 2015-03-11:
fixed