acfr/snark #97: view-points: output clicked points

junderwood, 2015-03-30:
The new points output feature outputs different numbers depending on camera angle, and the points bear no clear relationship to the input, e.g.:

echo -e "0,0,0\n1,1,1\n2,2,2" | view-points
... click, click, click... and observe output

andsok75, 2015-03-30:
Since the coordinates of the clicked points are inferred form pixel coordinates, they are not going to be precise. On my box I'm getting a precision of about 2% for this example, and the errors do depend on the viewing angle. For instance, when I click on the origin, I get the following for different viewing angles:
echo -e "0,0,0\n1,1,1\n2,2,2" | view-points
view-points: end of -; read 3 record(s)
view-points: reader no. 0 scene offset (0 0 0); scene radius: 10
0.0008422015234827995,0.003140207380056381,0.02548216842114925
-0.01628369837999344,0.009617313742637634,0.02226435393095016
-0.01628369837999344,0.009617313742637634,0.02226435393095016
-0.007217647973448038,-0.005692384205758572,0.005947224795818329

vlaskine, 2015-03-30:
@junderwood : is @andsok75 's explanation sufficient? if yes, could you please close this ticket; if no, what would be the desired behaviour?

it may not be possible to output the exact coordinates without a very substantial additional work, 2-3 weeks full-time - at least i don't see an easy way

junderwood, 2015-03-30:
Two things:
1) I get much bigger errors for this example. Following are the points I get out of view-points, instead of 0,0,0 1,1,1 and 2,2,2, I get:
3.540341138839722,3.540341138839722,-1.540341138839722
3.527428150177002,3.527184009552002,-1.566411137580872
7.381537914276123,7.381293773651123,-5.3621826171875

2) I assume this is a bug (above), but referring to Andrey's comment, assuming correct behaviour should be ~2% error. If ever we have a known inaccuracy, this should be more clearly highlighted, because these utilities are usually for scientific computing. It should be highlighted in the help, etc, and also to me or whoever is asking for the feature in the first place. I had no idea they weren't precise.

andsok75, 2015-03-31:
I reckon the precision will depend on your display resolution and possibly on the driver as well. If you wish to get precise coordinates, label-points is the only way at the moment.

I've adjusted help in view-points to make it clear that the output coordinates are approximate.

junderwood, 2015-03-31:
ok, thanks. But are you saying that the points in my comment above are within expected tolerances? If so, I'm tempted to suggest we remove the feature altogether.... ?

andsok75, 2015-03-31:
The coordinates in your example seem way off. Definitely not what I would expect.

junderwood, 2015-04-01:
To get the error I just use:
echo -e "0,0,0\n1,1,1\n2,2,2" | view-points and right click.

Note: it is failing on a virtual box - whereas it does work for me on windows.
Interestingly, it also works if I ssh -X to ladybird from my virtual box!

I guess we have to assume it's a vmware problem and not look further until it shows up on a real linux box. Kind of upsetting though (that the vmware isn't perfect)