acfr/snark #98: label-points: setting axis places it right in front of camera, instead of on the datum

junderwood, 2015-03-30:
echo -e "0,0,0,0\n1,1,1,0\n2,2,2,0" > test.csv
label-points "test.csv;fields=x,y,z,id"
.... then try to set location of camera axis (hold right button, while double clicking left over a point).
The axis should now be at the point you clicked on, but lately, the axis appears right in front of the camera (very large, and certainly in the wrong place).
I couldn't find a way to set the axis onto a point, in the way that it used to work.

vlaskine, 2015-03-30:
at which point did it stop working as you would expect?

junderwood, 2015-03-30:
I'm reporting it as soon as I've noticed it... but I haven't used it in a while, so I couldn't easily say.
I think it worked ok about 1 to 2 months ago.