acfr/snark #99: label-points: nearly impossible to select points

junderwood, 2015-03-30:
(likely related to axis issue I just posted)
I find it nearly impossible to select a point in label points, using any of the tools.

vlaskine, 2015-03-30:
did it work for you before? when did it start to fail?

junderwood, 2015-03-30:
I'm reporting it as soon as I've noticed it... but I haven't used it in a while, so I couldn't easily say.
I think it worked ok about 1 to 2 months ago.

vlaskine, 2015-03-30:
what is the priority of this bug? it may take several days to fix

we experience quite acute work shortage now: too many projects start activating at the same time

andrew-hill, 2015-03-30:
its always been pretty hard I reckon, and I think gets harder as your screen resolution increases (from memory its a 1-px tolerance for the point). A quick-and-dirty fix might be to just bump that tolerance to 2 or 3...

vlaskine, 2015-03-30:
thank you @andrew-hill , probably it might be good to make tolerance configurable from gui

junderwood, 2015-03-30:
sounds like a good option. it can be on the backlog for now and we'll prioritise during the next planning