acfr/snark #101: compile error LLVM + math/spherical_geometry/ellipsoid.h

andrew-hill, 2015-05-12:
This is a case of GCC being more permissive than LLVM (and apparently the C++11 standard, according to http://stackoverflow.com/q/16759025/343786 )

in ellipsoid.h, `ellipsoid::circle::arc` is defined within the `ellipsoid` struct, but apparently should be defined at the namespace level (outside of the `ellipsoid` definition)

still compiles in Ubuntu 13.04 with gcc so presumably this is OK to check in? (just want to check first as its not totally trivial and maybe slightly less elegant looking than previous code...)

```diff
diff --git a/math/spherical_geometry/ellipsoid.h b/math/spherical_geometry/ellipsoid.h
index 7970c6d..20f4932 100644
--- a/math/spherical_geometry/ellipsoid.h
+++ b/math/spherical_geometry/ellipsoid.h
@@ -73,29 +73,30 @@ struct ellipsoid
         std::vector< coordinates > discretize( ellipsoid &ellipsoid, const boost::optional< double > &resolution, const boost::optional< unsigned int > &circle_size ) const;
     };
 
-    //circular arc
-    struct circle::arc
-    {
-        /// circle of which this is an arc
-        ellipsoid::circle circle;
-        /// begin bearing
-        double begin;
-        /// end bearing
-        double end;
-        arc() {}
-        arc( const ellipsoid::circle &circle, double begin, double end ) : circle( circle ), begin( begin ), end( end ) {}
-        std::vector< coordinates > discretize( ellipsoid &ellipsoid, const boost::optional< double > &resolution, const boost::optional< unsigned int > &circle_size ) const;
-        
-        //heading degree from north; east is positive 90
-        static double bearing( double heading )
-        {
-            //bearing standard geometric radian; ie east=0,north=pi/2,south=-pi/2
-            return ( heading * -1 ) + ( M_PI / 2 );
-        }
-    };
+};
 
+//circular arc
+struct ellipsoid::circle::arc
+{
+    /// circle of which this is an arc
+    ellipsoid::circle circle;
+    /// begin bearing
+    double begin;
+    /// end bearing
+    double end;
+    arc() {}
+    arc( const ellipsoid::circle &circle, double begin, double end ) : circle( circle ), begin( begin ), end( end ) {}
+    std::vector< coordinates > discretize( ellipsoid &ellipsoid, const boost::optional< double > &resolution, const boost::optional< unsigned int > &circle_size ) const;
+
+    //heading degree from north; east is positive 90
+    static double bearing( double heading )
+    {
+        //bearing standard geometric radian; ie east=0,north=pi/2,south=-pi/2
+        return ( heading * -1 ) + ( M_PI / 2 );
+    }
 };

+
 }
 } // namespace snark { namespace spherical {
```

vlaskine, 2015-05-12:
@andrew-hill : sounds good, thank you for fixing it

could you please push the change and then close the ticket

andrew-hill, 2015-05-12:
fixed in 85cf205ef20e252a539d13003e3f14a7f104b757

closed 2015-05-12 by andrew-hill