acfr/snark #102: Cannot build snark with boost 1.58

vendall, 2015-06-10:
Hello,

I can't seem to build snark with boost 1.58. Previously I was able to downgrade it to 1.56, but due to other dependency issues I can no longer do this. 

Error is as follows:

[ 18%] Building CXX object graphics/applications/CMakeFiles/label-points.dir/label_points/IdEdit.cpp.o
/home/awen4053/src/snark/graphics/applications/label_points/IdEdit.cpp: In member function ‘void snark::graphics::View::IdEdit::handleReturnPressed()’:
/home/awen4053/src/snark/graphics/applications/label_points/IdEdit.cpp:81:55: warning: ‘*((void*)& maxId +4)’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         if( s == "new" ) { id = maxId ? *maxId + 1 : 0; }
                                                       ^
/home/awen4053/src/snark/graphics/applications/label_points/IdEdit.cpp:78:24: warning: ‘*((void*)& minId +4)’ may be used uninitialized in this function [-Wmaybe-uninitialized]
             if( !minId || m_viewer->datasets()[i]->partitions().begin()->first < *minId ) { minId = m_viewer->datasets()[i]->partitions().begin()->first; }
                        ^
Linking CXX executable ../../bin/label-points
/usr/bin/ld: warning: libboost_thread.so.1.56.0, needed by /usr/local/CMakeFiles/../lib/libcomma_application.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libboost_filesystem.so.1.56.0, needed by /usr/local/CMakeFiles/../lib/libcomma_application.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libboost_system.so.1.56.0, needed by /usr/local/CMakeFiles/../lib/libcomma_application.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libboost_date_time.so.1.56.0, needed by /usr/local/CMakeFiles/../lib/libcomma_application.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libboost_program_options.so.1.56.0, needed by /usr/local/CMakeFiles/../lib/libcomma_application.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libboost_regex.so.1.56.0, needed by /usr/local/CMakeFiles/../lib/libcomma_application.so, not found (try using -rpath or -rpath-link)
CMakeFiles/label-points.dir/label-points.cpp.o: In function `comma::name_value::map::init(comma::name_value::impl::options const&)':
/home/awen4053/src/snark/../comma/name_value/../name_value/map.h:107: undefined reference to `comma::split_escaped(std::string const&, char, char const*, char)'
collect2: error: ld returned 1 exit status
graphics/applications/CMakeFiles/label-points.dir/build.make:443: recipe for target 'bin/label-points' failed
make[2]: *** [bin/label-points] Error 1
CMakeFiles/Makefile2:156: recipe for target 'graphics/applications/CMakeFiles/label-points.dir/all' failed
make[1]: *** [graphics/applications/CMakeFiles/label-points.dir/all] Error 2
Makefile:146: recipe for target 'all' failed
make: *** [all] Error 2

Cheers,
Alex.

vlaskine, 2015-06-10:
are those errors from building against boost.1.58?
because your cmake seems to still point to boost.1.56

vendall, 2015-06-10:
Yes, both comma and snark were built with boost version 1.58.0-2 installed...

junderwood, 2015-06-10:
I took a look on Alex's machine - couldn't see where it was picking up the boost 1.56 reference. We completely rebuilt comma and snark from scratch and both cmakes were pointing to 1.58. Might be worth a quick look on his machine (he sits near you, just opposite the fish tank next to Suchet)

vlaskine, 2015-06-10:
although it may be a bug in cmake, it also may be fairly difficult, if possible at all to fix the build, if both boost.1.56 and 1.58 are installed on the same machine (especially if one is installed in /usr/lib and the other in /usr/local/lib)

is it the case? if yes, would it be an option to remove boost.1.56?

if yes, could you please try to remove 1.56, rebuild snark, and let me know whether it works

if you have to keep two versions of boost on the same machine, please let me know: i will come down to take a look

thank you!

vendall, 2015-06-10:
Thanks for your help, guys!

I'll do a more thorough check to make sure I don't also have 1.56 installed. I'll let you know how I go.

Alex.

nmccouat, 2015-06-11:
Hey guys, it might be worth trying to explicitly restrict boost - we use the following on our rhel6 machines to point it to our own version irrespective of installed / system versions. I'd be interested to know if this fixes the problem without uninstalling 1.56 - come and see me if you have any questions 

# Boost library
set( Boost_NO_BOOST_CMAKE        ON  )
set( Boost_USE_MULTITHREADED     ON  )
set( Boost_USE_STATIC_RUNTIME    OFF )
set( Boost_NO_SYSTEM_PATHS       ON  )

# Optionally set the compiler explicitly in case FindBoost finds a different gcc
# Our Boost build is gcc 4.4 only.
#if( CMAKE_COMPILER_IS_GNUCXX )
#set( Boost_COMPILER "-gcc44" )
#message( "Setting Boost_COMPILER" )
#endif()

set( BOOST_ROOT <directory where your 1.58 installation lives> )

vendall, 2015-06-11:
Hi guys,

I haven't had any luck with this yet. I definitely do not have 1.56 installed, only 1.58. I also tried rebuilding qt3d to see whether that would make any difference....

I have no idea why the linker is even looking for version 1.56 in the first place.

Any further suggestions?

Cheers,
Alex.

vlaskine, 2015-06-11:
I will go down to your desk to take a look. Where do you sit? What time is good for you after 12 on Friday?

vendall, 2015-06-11:
I won't be in most of Friday, unfortunately. Any time after 10am on Monday works for me....

vlaskine, 2015-06-11:
ok; i can come by on monday, e.g. at 12:30

where are you sitting, though? write to my university e-mail

vendall, 2015-06-18:
Hi everyone,

Seva determined the issue: It turns out that in a previous build and install of comma, I had included shared libraries. In the latest build, I didn't do this, and so only the static libraries got built and replaced, keeping the old shared libraries, which presumably pointed to the boost libraries on the machine at the time of their build (1.56). 

Manually deleting the old libraries fixed the issue, though rebuilding comma with shared libraries would probably have the done the trick too (overwriting the old shared library files).

Thanks for all your help - problem solved!

Cheers,
Alex.

closed 2015-06-18 by vendall