acfr/snark #103: cmake 2.8.2 (shrimp.littleboard) fails on terrain/CMakeLists.txt

andsok75, 2015-07-02:
cmake 2.8.2 (shrimp.littleboard) fails on terrain/CMakeLists.txt. 
TARGET_OBJECTS is a new syntax not supported by this version of cmake.

-- auto detected system manager: init
-- Configuring done
CMake Error in terrain/CMakeLists.txt:
  Cannot find source file "$<TARGET_OBJECTS:snark_terrain_dem_srtm>".  Tried
  extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp .hxx .in
  .txx

CMake Error in terrain/CMakeLists.txt:
  Cannot find source file "$<TARGET_OBJECTS:snark_terrain_dem>".  Tried
  extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp .hxx .in
  .txx

CMake Error in terrain/dem/CMakeLists.txt:
  Cannot find source file "OBJECT".  Tried extensions .c .C .c++ .cc .cpp
  .cxx .m .M .mm .h .hh .h++ .hm .hpp .hxx .in .txx

CMake Error in terrain/dem/srtm/CMakeLists.txt:
  Cannot find source file "OBJECT".  Tried extensions .c .C .c++ .cc .cpp
  .cxx .m .M .mm .h .hh .h++ .hm .hpp .hxx .in .txx

-- Build files have been written to: /home/segway/code/build/snark

vlaskine, 2015-07-02:
@andsok75 : any chance to upgrade cmake on segways? cmake already is 3.3.0

if it is too troublesome, could you please make terrain optional, off by default?

thank you!

andsok75, 2015-07-03:
I made terrain optional.

closed 2015-07-03 by andsok75