acfr/snark #104: Qt5.5 appears to have split Qt3D into multiple modules

andrew-hill, 2015-07-13:
**Good news**: Qt5.5 release notes indicate Qt3D is now (finally) included in the pre-built binaries, so it seems to be a legit component of Qt!

**Bad news**: It's been separated into a few submodules, so instead of just Qt3D there is now
* Qt3DCore
* Qt3DInput
* Qt3DRenderer

These two also start with Qt3D but I'm pretty sure Qt3DQuick was separate all along:
* Qt3DQuick
* Qt3DQuickRenderer

So this rather complicates the ability to build against Qt3D into three categories:
* Qt4
* Qt5 (up to 5.4)
* Qt5 (5.5+)

I've not actually found anything yet about this split (googling for Qt3DCore produces very few results), but assuming its the new structure going-forward, the question is whether to support all three of the above, or just the 'latest' state of Qt4 and Qt5.

*Essentially, where we currently test for Qt4 vs Qt5, should I just switch the current Qt5 behaviour to be compatible with 5.5 only? or do we want to have 3 options (4, 5, 5.5+)?*

vlaskine, 2015-07-13:
great news!

i would be careful about disrupting the support of qt4 for now

those who have qt5, can as well get qt5.5

is the required change extensive?

andrew-hill, 2015-07-13:
bad news... the version of Qt3D included in Qt5.5 is Qt3D 2.0 (previously we've been using 1.0 or patches upon 1.0)

so Qt5.5 onwards will include Qt3D 2, which is a complete rewrite of Qt3D—it sounds like it might even be a very design philosophy, not just small adjustments to the API

see comments here: http://www.kdab.com/qt3d-technology-preview-released-qt-5-5-0/

vlaskine, 2015-10-22:
somehow, with compilation and runtime warnings, view-points and label-points happily rebuild with qt5.5

so far, i have not spotted any functionality broken (but i have not tried everything)

thus, i close this ticket; please report any specific qt5.5-related failures in separate tickets

andrew-hill, 2015-11-11:
so i'm not exactly sure how this could have worked, unless the Mac and Linux qt5.5 contain different sources...

e.g.
`snark/graphics/qt3d/coordinates.h` includes `Qt3D/qgltexture2d.h`, which does not exist in my Qt5.5 installation

vlaskine, 2015-11-11:
strange indeed; i am not sure, but i saw it working on linux and certainly without installing qt3d

would you mind to try to do something like?

#ifndef APPLE
#include <Qt3D/qgltexture2d.h>
#endif

andrew-hill, 2015-11-11:
`coordinates.h` includes:

    #include <Qt3D/qgltexture2d.h>
    #include <Qt3D/qglscenenode.h>                                                                                                                                                 
    #include <Qt3D/qglbuilder.h>
    #include <Qt3D/qglpainter.h>
    #include <Qt3D/qglcylinder.h>

i can #ifdef them all out (none of these files exist anywhere in the 5.5 source from git), but it still won't build due to lack of a definition of QGLSceneNode

In file included from /Users/andrew/Development/snark/graphics/qt3d/coordinates.cpp:33:
/Users/andrew/Development/snark/graphics/qt3d/coordinates.h:49:5: error: unknown type name 'QGLSceneNode'
    QGLSceneNode* node() const;

(from a quick look this applies to most of the Qt3D includes and classes used in view-points and label-points)

i've checked the entire 5.5 branch on git for the string "QGLSceneNode" and it only exists in a text file that's the log output from some test (which seems out of date), so I'm actually not sure how you could have built view-points and label-points against Qt5.5

there's not even a Qt3D module for CMake to find anymore when it looks for Qt, so it shouldn't have even got past the cmake configure with Qt5.5

vlaskine, 2015-11-11:
hm... this is not a good news; i am pretty sure we built it on a computer that did not have qt3d installed and had qt.5.5

it built, complained a lot during the runtime, but eventually worked

moreover, let me ask @junderwood : i think he recently has installed comma and snark on a fresh linux box and, to my (likely false) memory did not need to install qt3d to have view-points working; @junderwood, could you comment?

junderwood, 2015-11-11:
I wanted to build snark against qt5.5 but was unable to easily install qt5.5... No apt get package. I ended up with 5.2 I think. Then, following the snark wiki instructions verbatim I was unable to get snark cmake to recognise qt3d with qt5.2. So following advice from navid or someone, I had to go with qt4 which worked. I have no regrets since then.

I made minor changes to the wiki, but  from my own experience the wiki should not even mention qt5.

vlaskine, 2015-11-11:
as i said, i have memories of installing qt.5 without qt3d (i think on Mitch's computer)

i guess we will have to reopen this ticket and prioritize fixing view-points and label-points for qt.5.5, which may be weeks of work