acfr/snark #105: gige-cat: wrong error when framerate and exposure time conflict

andrew-hill, 2015-07-30:
If the FrameRate (in Hz) and ExposureValue (in µs) conflict, such that the exposure times take longer than the time between frames:

    FrameRate > 1 / (ExposureValue / 1e6)

then you get a misleading error from gige-cat suggesting the MTU is too small

This can happen in two ways. Either **Manual** exposure mode, directly configuring them to be of range of one another (e.g. `FrameRate=30;ExposureValue=500000`), or **Auto** exposure mode with an `ExposureAutoMax` value exceeding the time between frames. The latter will not fail immediately, so can be easily missed and only discovered much later (so its hard to notice the correlation between the FrameRate/Exposure settings and the failure), it only fails/exits when the image gets dark enough to push the ExposureValue past some threshold where gige-cat will exit.

I'm not entirely sure if the fault lies in the camera or in gige-cat's logic, but looking at gige.cpp i think its only checking the expected FrameRate value, and exiting if you have too many missed frames, without considering that it could be starved by a long ExposureValue.

snark/sensors/cameras/gige/gige.cpp:246

            PvAttrFloat32Get( handle_, "FrameRate", &frameRate);
            timeOut_ = 1000 / frameRate;
            timeOut_ *= timeOutFactor;

Some suggestions

* warn (or exit with error) up front if either
 * ExposureValue / 1e6 > 1/FrameRate
 * ExposureMode=Auto && ExposureAutoMax / 1e6 > 1/FrameRate
* set `timeOut` based on max of
 * 1000 / frameRate
 * 1000 * ExposureValue
 * iff ExposureMode=Auto, 1000 * ExposureAutoMax

I *think* this is a pretty easy fix, so can probably look at doing it myself in the next few days unless someone else has better ideas...

I guess there's some ambiguity about what the expected behaviour should be if you have (intentionally or unintentionally) configured the camera to (possibly) generate frames at a lower rate than FrameRate (i.e. is it "ideally X fps but drop down as necessary" or "i totally forgot to set sensible exposure settings and gige-cat should have crashed up-front so I know to fix it"

Should it crash/exit up front? or just warn and allow a larger timeout? or should you have to specify an extra option to allow this degraded-fps mode?

vlaskine, 2015-07-30:
@andrew-hill : I think the checks should happen upfront and if exposure conflicts with frame rate, gige-cat should exit with a corresponding error message. It would be terrific, if you could make that change.

andrew-hill, 2015-07-30:
what about the mode where you want max possible FPS, but still get well-lit images?

i.e. you set "FrameRate=30;ExposureAutoMax=500000" to give you 2-30 fps depending on lighting conditions

vlaskine, 2015-07-30:
does it still exit with error in this case? ("FrameRate=30;ExposureAutoMax=500000" to give you 2-30 fps)

andrew-hill, 2015-07-30:
the short answer is "kind of" — in its current state, it'll exit if/when the frame gets dark enough to push the ExposureValue to something like 3x the FrameRate period, so you do 10-30fps but not 2-30fps

vlaskine, 2015-07-30:
would it be an option for now to just print an adequate error message on exit?

the next step would be to introduce an option (something like --permissive or better --discard-underexposed) that would allow to skip this kind of underexposed frames

would it solve the problem?

junderwood, 2015-07-30:
I think for any scenario where future lighting conditions might cause an exit, the exit should be up front immediately on start rather than a warning. However for any case other than that, we should be permissive and give warnings. I don't mind if there's a permissive setting if that seems to make sense to people, but we never should allow a random future failure in any case.

andrew-hill, 2015-07-31:
Agreed, and I can fix the problem of exiting later on quite easily I think.

So the question isn't whether to exit immediately or exit later—it's whether to exit immediately OR accept that the frame rate might drop below the requested frame rate (but will still continue operating), if the camera is configured in a way that it may not meet the requested frame rate.

The latter is more true to the actual camera's behaviour (the exit isn't caused by the camera, it does keep working), while the former is more true to protecting the user from (possible/assumed) unintentional mistake (but note that its entirely possible for this configuration to be intentional too).

The default behaviour could be either of the above (exit immediately or just keep going at whatever frame rate the camera produces based on the config), with a warning and/or option to switch this behaviour.

vlaskine, 2015-07-31:
i would suggest to add an option (e.g. --ignore-underexposed-frames, if it conveys the meaning correctly); if the option is not there, exit

andrew-hill, 2015-07-31:
just to be clear, the reduced frame-rate mode isn't ignoring or discarding frames, nor are they being underexposed — quite the opposite in fact — they are taking longer to expose so that they're sufficiently exposed (e.g. half a second exposure time means only 2 frames get exposed and sent back by the camera per second, not that 28 failed)

so I want to make sure we're not confusing two separate issues, e.g. it would be a wholly separate issue if 28 of 30 frames per second are being lost/discarded somehow, and in that case, gige-cat wouldn't be faithfully operating as the camera is operating... but that's another case

the case i'm talking about, gige-cat is functioning exactly as per the sensor design—no frames are lost/discarded, there's just less data per second than one *might* be expecting

as for options... the existing behaviour is that the frame rate is enforced between the requested FrameRate (the camera will max out at that value) and 1/3 of FrameRate (gige-cat will exit below this value), so we could offer settings such as:

* `--allow-auto-redcuced-frame-rate` (figure out worst case from auto settings, and probably still apply some factor of safety)
* `--frame-timeout=500` (manually specify absolute timeout in ms, instead of calculating from settings with arbitrary factor of safety)
* `--min-frame-rate=2` (likewise but in Hz)

and in all cases, exit immediately if it's possible for a frame exposure time to exceed the calculated timeout (with at least a small factor of safety built-in)

**but** i'm still not sure that it makes sense to ignore the ExposureAutoMax setting to figure out min frame rate **by default**—we're *already* arbitrarily allowing reduced frame rates down to 1/3 requested. In my mind, the exit is there to indicate **sensor failure**—as it states the MTU is probably set wrong, you're not getting any data, so don't let the service run—rather than to enforce a minimum frame-rate. And I assume the timeout is 3x expected frame period just so it won't trigger in normal operation, not because 1/3 frame rate is a useful minimum frame-rate.

vlaskine, 2015-07-31:
@andrew-hill : if you see a way to make the default behaviour more consistent, please go for it (also agree on all the options you suggested)

andrew-hill, 2015-07-31:
updates committed, i think this now makes more sense and keeps config arguments to bare minimum... timeout is calculated from camera settings, and you can override it if need be

closed 2015-07-31 by andrew-hill

vlaskine, 2015-07-31:
excellent, thank you!

junderwood, 2015-07-31:
Thanks Andrew!