acfr/snark #107: undefined reference to comma::verbose

andsok75, 2015-10-20:
Compilation fails on points-calc.cpp with the following error:

CMakeFiles/points-calc.dir/points-calc.cpp.o: In function `operator<< <char [7]>':
/home/andrey/src/snark/../comma/application/verbose.h:67: undefined reference to `comma::verbose'
CMakeFiles/points-calc.dir/points-calc.cpp.o: In function `plane_intersection::run(comma::csv::options const&, boost::optional<std::string>, bool)':
/home/andrey/src/snark/point_cloud/applications/detail/plane-intersection.h:148: undefined reference to `comma::verbose_t::operator<<(std::ostream& (*)(std::ostream&))'
/home/andrey/src/snark/point_cloud/applications/detail/plane-intersection.h:151: undefined reference to `comma::verbose_t::operator<<(std::ostream& (*)(std::ostream&))'
collect2: error: ld returned 1 exit status
make[2]: *** [bin/points-calc] Error 1
make[1]: *** [point_cloud/applications/CMakeFiles/points-calc.dir/all] Error 2
make: *** [all] Error 2

vlaskine, 2015-10-20:
it seems to work everywhere except your machine

something to sort out between you and @navidp ?

closed 2015-10-20 by vlaskine

andsok75, 2015-10-20:
Deleting

/usr/local/lib/libcomma_*
/usr/local/include/comma/
/usr/local/lib/libsnark_*
/usr/local/include/snark/

and rebuilding has fixed the problem.