acfr/snark #108: building snark + qt5 fail

junderwood, 2015-10-28:
I followed this:
https://github.com/acfr/snark/wiki/Snark-%28and-comma%29-for-dummies,-on-ubuntu-debian
On a fresh install of Ubuntu 14.04 LTS

1)
sudo apt-get install libqt5-dev : libqt5-dev apparently doesn't exist
so I installed qt5-default (google told me to):
sudo apt-get install qt5-default
I couldn't find a way to explicitly install qt5.5like this, and I ended up with qt5.2

2)
I followed the instructions to install qt3d (verbatim, even where it says "qt4")
https://github.com/acfr/snark/wiki/Snark-%28and-comma%29-for-dummies,-on-ubuntu-debian
...the teapot displayed but had some bad visual tearing.

3)
cmake of snark fails to find qt3d. Qt53D_DIR not found.

junderwood, 2015-10-28:
I also tried:
sudo apt-get install libqt53d5
but that didn't help, snark cmake still complains

navidp, 2015-10-28:
set the "snark_QT_VERSION" to 4 in ccmake .

vlaskine, 2015-10-28:
@junderwood : once it eventually gets resolved, could you please correct snark/wiki? thank you!

please let me know, if you would like me to come over and help

junderwood, 2015-10-28:
This worked. However, it was hard to 'tweak ' the wiki, as my solution was roundabout. I used apt-get to get qt5-default, but then set snark_QT_VERSION to "4"... maybe my system had a version of qt4 already perhaps...?

I think it'd be good to update the doc to say how to actually get qt5.5, which I still don't know. qt5-default for me was 5.2

Conclusion: I now have view-points, so I don't need to follow up further. I made the following changes to the wiki:
sudo apt-get install qt5-default (instead of libqt5-dev which doesn't exist)
I added the following vague comment:
** Note: Set snark cmake variable 'snark_QT_VERSION' appropriately

closed 2015-10-28 by junderwood

andsok75, 2015-10-28:
It probably makes more sense to change change the instructions to "instal libqt4-dev", rather than qt5-default, since according to what you said qt5 does not work.

junderwood, 2015-10-28:
Probably. Turns out, it's all a bit broken. teapot and viewpoints have the same rendering problem... rendering them useless. I'm trying to cleanse the system of qt5. Wow... it was a brand new system this morning and it's already screwed.

junderwood, 2015-10-28:
Solved: it was a graphics driver issue. 
Search -> Additional drivers -> Select "Using NVIDIA binary driver - version 340.93 from nvidia-340-updates.