acfr/snark #114: csv-plot build errors with QT5

spotiris, 2016-04-13:
Snark builds fine without the graphics module, but when enabled gives build errors:
`
/usr/include/qwt/qwt_global.h:13:21: fatal error: qglobal.h: No such file or directory
 #include <qglobal.h>
`
When QT version 5.2.1 is installed with QWT version 6 installed (libqwt6).

vlaskine, 2016-08-19:
made it building with qt5, but it still segfaults
whoever reads it, please talk to me, if you really need to use csv-plot, since we rather phase it out and implement a nice stream-based wrapper around pyplot

closed 2016-08-19 by vlaskine

david-jennings, 2016-08-26:
Qt5 requires Qwt 6.1 or greater. Cmake now enforces this requirement. Ubuntu 14.04 ships Qwt 6.0 but Ubuntu 16.04 has the correct version.