acfr/snark #116: Missing check for contrib.hpp in CMakeLists.txt

josch, 2017-01-18:
I'm able to successfully configure the project but then during building I get:

    [ 34%] Building CXX object imaging/CMakeFiles/snark_imaging.dir/cv_mat/filters.cpp.o
    /home/3dtk/snark/imaging/cv_mat/filters.cpp:58:10: fatal error: 'opencv2/contrib/contrib.hpp' file not found
    #include <opencv2/contrib/contrib.hpp>
             ^
    1 error generated.
    imaging/CMakeFiles/snark_imaging.dir/build.make:134: recipe for target 'imaging/CMakeFiles/snark_imaging.dir/cv_mat/filters.cpp.o' failed
    make[2]: *** [imaging/CMakeFiles/snark_imaging.dir/cv_mat/filters.cpp.o] Error 1
    CMakeFiles/Makefile2:1065: recipe for target 'imaging/CMakeFiles/snark_imaging.dir/all' failed
    make[1]: *** [imaging/CMakeFiles/snark_imaging.dir/all] Error 2
    Makefile:160: recipe for target 'all' failed
    make: *** [all] Error 2

So apparently snark needs opencv's contrib.hpp but does not check for its existance during configuration.

vlaskine, 2017-01-18:
you probably are using opencv 3, which snark does not support yet

i have explicitly defined in cmake that version 2 is required; if have both opencv 2 and opencv 3 installed, you could pull snark and try to rebuild; let me know if it works; i close the ticket for now, because there is another ticket: add opencv 3 support

closed 2017-01-18 by vlaskine

josch, 2017-01-19:
You are right. I was in the situation to have both installed. I also ultimately stumbled across that when snark tried to link to the wrong opencv library versions and thus wasn't able to find certain symbols. After having downgraded to opencv2 everything works again. I'm sure you are taking care of the `contrib.hpp` problem as part of your opencv3 conversion, so I'm leaving this issue closed. Thanks!