acfr/snark #117: non-constant-expression cannot be narrowed from type 'int' to 'unsigned int' in initializer list

josch, 2017-01-18:
Hi,

when I try building snark with clang version 3.8.1-16 and `USE_CXX11` set to `ON` then I get multiple of the following errors:

    non-constant-expression cannot be narrowed from type 'int' to 'unsigned int' in initializer list

Mostly these are instances where you try to store a signed int as an unsigned int. For example in `point_cloud/applications/points-to-voxels.cpp` line 228.

vlaskine, 2017-01-18:
Hi,

We never have tried to compile with clang. (I am surprised it is so restrictive.) Please post your compilation errors and I will see what I can do.

Meanwhile, as another quick option for you, add a compiler flag in cmake that will make signed to unsigned conversions less restrictive (I am not sure what option it is in clang).

Thank you

josch, 2017-01-19:
My issue is not that I cannot compile snark. I just switch back to gcc or add `-Wnoc++11-narrowing` to my `CMAKE_CXX_FLAGS` and I can compile it again. I just thought you might care about these warnings and thus opened this issue.

Unfortunately I also do not have a complete list for you because not only are the lists very long but after having fixed the problem in one compilation unit, it pops up in the next. Here are the errors from the frist occurence:

    /home/3dtk/snark/point_cloud/applications/points-to-voxels.cpp:228:76: error: non-constant-expression cannot be narrowed
          from type 'unsigned int' to 'int' in initializer list [-Wc++11-narrowing]
      ...snark::voxel_map< centroid, 3 >::index_type begin = {{ it->first[0] - neighbourhood_radius, it->first[1] - neigh...
                                                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    /home/3dtk/snark/point_cloud/applications/points-to-voxels.cpp:228:76: note: insert an explicit cast to silence this
          issue
      ...snark::voxel_map< centroid, 3 >::index_type begin = {{ it->first[0] - neighbourhood_radius, it->first[1] - neigh...
                                                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                static_cast<int>(                  )
    /home/3dtk/snark/point_cloud/applications/points-to-voxels.cpp:228:113: error: non-constant-expression cannot be
          narrowed from type 'unsigned int' to 'int' in initializer list [-Wc++11-narrowing]
      ...3 >::index_type begin = {{ it->first[0] - neighbourhood_radius, it->first[1] - neighbourhood_radius, it->first[2...
                                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    /home/3dtk/snark/point_cloud/applications/points-to-voxels.cpp:228:113: note: insert an explicit cast to silence this
          issue
      ...3 >::index_type begin = {{ it->first[0] - neighbourhood_radius, it->first[1] - neighbourhood_radius, it->first[2...
                                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                         static_cast<int>(                  )
    /home/3dtk/snark/point_cloud/applications/points-to-voxels.cpp:228:150: error: non-constant-expression cannot be
          narrowed from type 'unsigned int' to 'int' in initializer list [-Wc++11-narrowing]
      ...- neighbourhood_radius, it->first[1] - neighbourhood_radius, it->first[2] - neighbourhood_radius }};
                                                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    /home/3dtk/snark/point_cloud/applications/points-to-voxels.cpp:228:150: note: insert an explicit cast to silence this
          issue
      ...- neighbourhood_radius, it->first[1] - neighbourhood_radius, it->first[2] - neighbourhood_radius }};
                                                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                      static_cast<int>(                  )
    /home/3dtk/snark/point_cloud/applications/points-to-voxels.cpp:229:74: error: non-constant-expression cannot be narrowed
          from type 'unsigned int' to 'int' in initializer list [-Wc++11-narrowing]
      ...snark::voxel_map< centroid, 3 >::index_type end = {{ it->first[0] + neighbourhood_radius + 1, it->first[1] + nei...
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    /home/3dtk/snark/point_cloud/applications/points-to-voxels.cpp:229:74: note: insert an explicit cast to silence this
          issue
      ...snark::voxel_map< centroid, 3 >::index_type end = {{ it->first[0] + neighbourhood_radius + 1, it->first[1] + nei...
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                              static_cast<int>(                      )
    /home/3dtk/snark/point_cloud/applications/points-to-voxels.cpp:229:115: error: non-constant-expression cannot be
          narrowed from type 'unsigned int' to 'int' in initializer list [-Wc++11-narrowing]
      ...>::index_type end = {{ it->first[0] + neighbourhood_radius + 1, it->first[1] + neighbourhood_radius + 1, it->fir...
                                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    /home/3dtk/snark/point_cloud/applications/points-to-voxels.cpp:229:115: note: insert an explicit cast to silence this
          issue
      ...>::index_type end = {{ it->first[0] + neighbourhood_radius + 1, it->first[1] + neighbourhood_radius + 1, it->fir...
                                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                         static_cast<int>(                      )
    /home/3dtk/snark/point_cloud/applications/points-to-voxels.cpp:229:156: error: non-constant-expression cannot be
          narrowed from type 'unsigned int' to 'int' in initializer list [-Wc++11-narrowing]
      ...+ neighbourhood_radius + 1, it->first[1] + neighbourhood_radius + 1, it->first[2] + neighbourhood_radius + 1 }};
                                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...] pp:229:156: note: insert an explicit cast to silence this
          issue
      ...+ neighbourhood_radius + 1, it->first[1] + neighbourhood_radius + 1, it->first[2] + neighbourhood_radius + 1 }};
                                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                              static_cast<int>(                      )
    

Just close this issue if you don't care about the problem. :)

vlaskine, 2017-01-19:
I have added -Wnoc++11-narrowing for Clang compiler in both comma and snark. (The error on assignment of signed to unsigned seems really over the top.) Could you please pull and try a fresh build? Let me know, if your compilation still fails.

josch, 2017-01-19:
Thanks!

But we get more:

    /home/3dtk/snark/imaging/cv_mat/filters.cpp:2564:102: error: 'logger' is a private member of 'snark::cv_mat::log_impl_'
        template < typename K, typename V > static void visit( const K&, const snark::cv_mat::log_impl_::logger::indexer...
                                                                                                         ^
    /home/3dtk/snark/imaging/cv_mat/filters.cpp:647:15: note: declared private here
            class logger
                  ^
    /home/3dtk/snark/imaging/cv_mat/filters.cpp:2564:110: error: 'indexer' is a private member of
          'snark::cv_mat::log_impl_::logger'
      ...typename K, typename V > static void visit( const K&, const snark::cv_mat::log_impl_::logger::indexer& t, V& v )
                                                                                                       ^
    /home/3dtk/snark/imaging/cv_mat/filters.cpp:681:24: note: declared private here
                    struct indexer
                           ^
    /home/3dtk/snark/imaging/cv_mat/filters.cpp:2571:96: error: 'logger' is a private member of 'snark::cv_mat::log_impl_'
        template < typename K, typename V > static void visit( const K&, snark::cv_mat::log_impl_::logger::indexer& ...
                                                                                                   ^
    /home/3dtk/snark/imaging/cv_mat/filters.cpp:647:15: note: declared private here
            class logger
                  ^
    /home/3dtk/snark/imaging/cv_mat/filters.cpp:2571:104: error: 'indexer' is a private member of
          'snark::cv_mat::log_impl_::logger'
      ...< typename K, typename V > static void visit( const K&, snark::cv_mat::log_impl_::logger::indexer& t, V& v )
                                                                                                   ^
    /home/3dtk/snark/imaging/cv_mat/filters.cpp:681:24: note: declared private here
                    struct indexer
                           ^

vlaskine, 2017-01-19:
thanks, g++ is so much more permissive than clang

i have pushed the fix, could you please pull snark and try again

thanks for your patience

josch, 2017-01-19:
As far as I see it I have to thank *you* for your patience *and* for your very quick replies. :)

Pulling your changes and trying them out is no big deal, so don't mention it. Thanks for maintaining snark!

Here is your next batch of errors:

    /home/3dtk/snark/sensors/cameras/dc1394/dc1394.h:118:36: error: type 'const snark::camera::dc1394::config' does not
          provide a call operator
        dc1394( const config& config = config(), const strobe& strobe = strobe() );
                                       ^~~~~~
    /home/3dtk/snark/sensors/cameras/dc1394/dc1394.h:118:69: error: type 'const snark::camera::dc1394::strobe' does not
          provide a call operator
        dc1394( const config& config = config(), const strobe& strobe = strobe() );
                                                                        ^~~~~~

vlaskine, 2017-01-19:
fixed, please try to pull and build

josch, 2017-01-19:
And it went through! Thanks!

vlaskine, 2017-01-19:
great, thanks for telling us about the problems; let us know if anything else does not work for you

closed 2017-01-19 by vlaskine