acfr/snark #118: view-points only shows black

josch, 2017-01-24:
Hi,

I tred to run the examples from http://www.acfr.usyd.edu.au/papers/icra13-underwood-changedetection.shtml but I only get a window that is completely black apart from the text "centre of rotation: 0 0 0" in the lower right corner. On the terminal I see:

```
$ view-points 'additions.bin;binary=6d;fields=x,y,z;colour=green' 'additions.bin;binary=6d;fields=first,second;colour=yellow;shape=line' 'subtractions.bin;binary=6d;fields=x,y,z;colour=red' 'subtractions.bin;binary=6d;fields=first,second;colour=yellow;shape=line' 'georef1.bin;binary=3d;fields=x,y,z;colour=grey' 'georef2.bin;binary=3d;fields=x,y,z;colour=grey' --scene-radius=100 --no-stdin
QOpenGLShader::compile(Vertex): 0:1(10): error: GLSL 1.50 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

*** Problematic Vertex shader source code ***
#version 150
#define lowp
#define mediump
#define highp
#line 2
in vec4 vertex;
in vec4 color;
out vec4 vert_color;
uniform mat4 projection_matrix;
uniform mat4 mv_matrix;
void main() {
   vert_color = color;
   gl_Position = projection_matrix * mv_matrix * vertex;
}

***
QOpenGLShader::compile(Fragment): 0:1(10): error: GLSL 1.50 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

*** Problematic Fragment shader source code ***
#version 150
#define lowp
#define mediump
#define highp
#line 2
in highp vec4 vert_color;
out highp vec4 frag_color;
void main() {
   frag_color = clamp( vert_color, 0.0, 1.0 );
}

***
```

Maybe the terminal output hints to the reason why I don't see anything with `view-points`?

Or maybe I'm seeing something but it's just black points in front of a black background? ;)

Or maybe my invocation is wrong. I already had to figure out that I have to add `--no-stdin` to the `view-points` invocation which is not part of the call made by `detect-change.sh` (can that script be updated to work with the latest view-points?).

vlaskine, 2017-01-25:
there seems to be some incompatibility with your graphics card on your computer (view-points is used for 3D visualisation only)

1) which linux version are you using? (run: cat /proc/version)

1) which Qt version do you use? (run: run qmake --version)

2) does qt3d demo work on your machine? (look for qt3d/bin/teapot and run it; you should see a 3D model of a teapot)

3) if you can see teapot in the previous example; try to run the following command; you should see 8 large points arranged as vertices of a cube (from what you report most likely it won't work):

( echo 0,0,0 ; echo 0,0,1 ; echo 0,1,0 ; echo 0,1,1 ; echo 1,0,0 ; echo 1,0,1 ; echo 1,1,0 ; echo 1,1,1 ) | view-points --weight=10

tell me what you see

josch, 2017-01-25:
* OS: Debian unstable amd64
 * kernel: 4.8.0-1 
 * mesa: 13.0.2
 * qt: 5.7.1
 * gfx: intel hd graphics 5500

I couldn't find the teapot demo so instead I made a screenshot of the wireframe and basicshapes-cpp examples from the examples directory (source code [here](http://sources.debian.net/src/qt3d-opensource-src/5.7.1%2Bdfsg-1/examples/qt3d/)):

[image]
[image]

When I try your view-points invocation then I see nothing but blackness.

vlaskine, 2017-01-25:
view-points does not support Qt5.7 yet (work in progress due to a massive breaking change in Qt). 

If you are keen to have view-points working, you would need to install Qt4 (your should be able to have multiple Qt installations). Then you would need to get and build from source Qt3D. Then, rebuild snark.

The detailed step-by-step instructions (it's not that hard) are here: https://github.com/acfr/snark/wiki/Snark-for-dummies,-on-ubuntu-debian#Installation

josch, 2017-01-25:
Ooooh! Maybe there should be a cmake check or something about that? If view-points does not support Qt5.7, then I'd expect it not to build if the onl Qt version on the system is 5.7.

Thanks!

vlaskine, 2017-01-25:
I know; we have been working on it; it is such a massive change in qt that we have been a bit overwhelmed by it. If you still keen, please try to build with Qt4 and Qt3d

josch, 2017-01-25:
Alright, this issue can be closed then from my side unless you want to keep it open to track the Qt5.7 status with view-points.

junderwood, 2017-01-25:
Incidentally, if your ultimate goal is to use the change detection algorithm, you can also easily do it without viewpoints and use an alternate viewer of your choice...
It'd be a shame though because viewpoints is extremely useful for streams.

vlaskine, 2017-01-25:
Qt5.* support is a big fat item on our todo list; we won't forget, but it will take a while

closed 2017-01-25 by vlaskine