acfr/snark #119: points-to-voxels: fails when coordinates are too large

mikkelkh, 2017-07-13:
When working with large coordinates (e.g. UTM coordinates), points-to-voxels fails.
The same happens for points-to-voxel-indices.

Example:
echo -e "4608177800,0,0\n4608177801,0,0" | points-to-voxels --fields=x,y,z --resolution=0.1
returns:
-2147483648,0,0,4608177800.5,0,0,2

should return something like:
46081778010,0,0,4608177801,0,0,1
46081778000,0,0,4608177800,0,0,1

vlaskine, 2017-07-13:
points-to-voxels and points-to-voxel-indices use int32 for indices; it's a limitation; changing them to int64 would break backwards compatibility

on the other hand, you always can specify --origin in both utilities that would be closer to where your point cloud is; e.g:

> echo -e "4608177800,0,0\n4608177801,0,0" | points-to-voxels --fields=x,y,z --resolution=0.1
-2147483648,0,0,4608177800.5,0,0,2
> echo -e "4608177800,0,0\n4608177801,0,0" | points-to-voxels --fields=x,y,z --resolution=0.1 --origin 4600000000,0,0
81778010,0,0,4608177801,0,0,1
81778000,0,0,4608177800,0,0,1

having said that, 4608177800 is 4 million kilometres and those utilities were not meant for space exploration

mikkelkh, 2017-07-14:
That makes sense. Thanks for the clarification!

And thanks for the hint regarding --origin. That would actually make it work in space, as long as the point cloud does not span several planets :-)
Also, I can see that I should probably just have my coordinates specified in SI base units, as that would solve the problem altogether.

Thanks again, and sorry for the trouble.

closed 2017-07-14 by mikkelkh

vlaskine, 2017-07-14:
no problem, thanks Mikkel