acfr/snark #120: error: ‘accumulate’ is not a member of ‘std’

josch, 2017-07-23:
On Debian unstable with gcc 6.3.0 I get the following errors:

```
[ 37%] Building CXX object imaging/applications/CMakeFiles/cv-calc.dir/cv-calc.cpp.o
/home/3dtk/snark/imaging/applications/cv-calc.cpp: In member function ‘unsigned int snark::imaging::operations::grep::non_zero::count(const cv::Mat&) const’:
/home/3dtk/snark/imaging/applications/cv-calc.cpp:236:20: error: ‘accumulate’ is not a member of ‘std’
             return std::accumulate( counts.begin(), counts.end(), 0 );
                    ^~~
At global scope:
```

```
[ 35%] Building CXX object imaging/applications/CMakeFiles/image-color-calc.dir/image-color-calc.cpp.o
In file included from /home/3dtk/comma/install/CMakeFiles/../include/comma/csv/../application/command_line_options.h:43:0,
                 from /home/3dtk/comma/install/CMakeFiles/../include/comma/csv/options.h:36,
                 from /home/3dtk/snark/imaging/applications/../color/convert.h:35,
                 from /home/3dtk/snark/imaging/applications/image-color-calc.cpp:32:
/home/3dtk/snark/imaging/applications/image-color-calc.cpp: In function ‘snark::imaging::colorspace {anonymous}::get_colorspace_from_fields(const std::vector<std::__cxx11::basic_string<char> >&, const std::vector<snark::imaging::colorspace>&)’:
/home/3dtk/snark/imaging/applications/image-color-calc.cpp:179:63: error: ‘accumulate’ is not a member of ‘std’
                                                            << std::accumulate( std::next( spaces.begin() ), spaces.end(), std::string( spaces.front() )
                                                               ^
```

The following patch fixes the issue:

```
diff --git a/imaging/applications/cv-calc.cpp b/imaging/applications/cv-calc.cpp
index cc535071..98c0eb5b 100644
--- a/imaging/applications/cv-calc.cpp
+++ b/imaging/applications/cv-calc.cpp
@@ -39,6 +39,7 @@
 #include <comma/visiting/traits.h>
 #include "../../imaging/cv_mat/filters.h"
 #include "../../imaging/cv_mat/serialization.h"
+#include <numeric>
 
 const char* name = "cv-calc: ";
 
diff --git a/imaging/applications/image-color-calc.cpp b/imaging/applications/image-color-calc.cpp
index 8ca51ef7..aedb0beb 100644
--- a/imaging/applications/image-color-calc.cpp
+++ b/imaging/applications/image-color-calc.cpp
@@ -33,6 +33,7 @@
 #include "../color/traits.h"
 
 #include <comma/csv/stream.h>
+#include <numeric>
 
 // todo
 // - <operation> (for now only one operation: convert)  DONE
```

According to http://en.cppreference.com/w/cpp/algorithm/accumulate one has to include the `<numeric>` header to use `std::accumulate`.

vlaskine, 2017-07-23:
thank you very much; i have pushed the fix

closed 2017-07-23 by vlaskine

josch, 2017-07-23:
Thanks, but your recent commit only fixed `imaging/applications/cv-calc.cpp` but not `imaging/applications/image-color-calc.cpp`.

vlaskine, 2017-07-23:
sorry, pushed the fix