acfr/snark #124: points-join doesn't build when CUDA enabled

spotiris, 2017-08-30:
When building snark with USE_CUDA=1 points-join doesn't build.
I verified that it does build when USE_CUDA=0.

`[ 77%] Building CXX object point_cloud/applications/CMakeFiles/points-join.dir/points-join.cpp.o
/home/steven/src/snark/point_cloud/applications/points-join.cpp: In member function ‘boost::optional<std::pair<Eigen::Matrix<double, 3, 1>, double> > traits<Eigen::Matrix<double, 3, 1> >::voxel_t::nearest_to(const point_input&, unsigned int) const’:
/home/steven/src/snark/point_cloud/applications/points-join.cpp:223:218: error: no match for ‘operator-’ (operand types are ‘const Vector3d {aka const Eigen::Matrix<double, 3, 1>}’ and ‘const point_input’)
 t& rhs, unsigned int k ) const { return std::make_pair( records[k]->value, use_cuda ? buffer.out[k] : (records[k]->value - rhs ).squ
                                                                                                                          ^
In file included from /usr/include/eigen3/Eigen/src/Core/MatrixBase.h:126:0,
                 from /usr/include/eigen3/Eigen/Core:345,
                 from /home/steven/src/snark/point_cloud/applications/../../math/geometry/polygon.h:35,
                 from /home/steven/src/snark/point_cloud/applications/points-join.cpp:46:
/usr/include/eigen3/Eigen/src/Core/../plugins/CommonCwiseUnaryOps.h:50:1: note: candidate: const NegativeReturnType Eigen::MatrixBase<Derived>::operator-() const [with Derived = Eigen::Matrix<double, 3, 1>; Eigen::MatrixBase<Derived>::NegativeReturnType = Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<double>, const Eigen::Matrix<double, 3, 1> >; typename Eigen::internal::traits<T>::Scalar = double]
 operator-() const { return NegativeReturnType(derived()); }
 ^
/usr/include/eigen3/Eigen/src/Core/../plugins/CommonCwiseUnaryOps.h:50:1: note:   candidate expects 0 arguments, 1 provided
In file included from /usr/include/eigen3/Eigen/Core:62:0,
                 from /home/steven/src/snark/point_cloud/applications/../../math/geometry/polygon.h:35,
                 from /home/steven/src/snark/point_cloud/applications/points-join.cpp:46:
/usr/include/eigen3/Eigen/src/Core/../plugins/CommonCwiseBinaryOps.h:19:28: note: candidate: template<class OtherDerived> const Eigen::CwiseBinaryOp<Eigen::internal::scalar_difference_op<typename Eigen::internal::traits<T>::Scalar>, const Derived, const OtherDerived> Eigen::MatrixBase<Derived>::operator-(const Eigen::MatrixBase<OtherDerived>&) const [with OtherDerived = OtherDerived; Derived = Eigen::Matrix<double, 3, 1>]
 EIGEN_MAKE_CWISE_BINARY_OP(operator-,internal::scalar_difference_op)
                            ^
/usr/include/eigen3/Eigen/src/Core/util/Macros.h:832:4: note: in definition of macro ‘EIGEN_MAKE_CWISE_BINARY_OP’
   (METHOD)(const EIGEN_CURRENT_STORAGE_BASE_CLASS<OtherDerived> &other) const \
    ^
/usr/include/eigen3/Eigen/src/Core/../plugins/CommonCwiseBinaryOps.h:19:28: note:   template argument deduction/substitution failed:
 EIGEN_MAKE_CWISE_BINARY_OP(operator-,internal::scalar_difference_op)
                            ^
/usr/include/eigen3/Eigen/src/Core/util/Macros.h:832:4: note: in definition of macro ‘EIGEN_MAKE_CWISE_BINARY_OP’
   (METHOD)(const EIGEN_CURRENT_STORAGE_BASE_CLASS<OtherDerived> &other) const \
    ^
/home/steven/src/snark/point_cloud/applications/points-join.cpp:223:220: note:   ‘const point_input’ is not derived from ‘const Eigen::MatrixBase<Derived>’
  rhs, unsigned int k ) const { return std::make_pair( records[k]->value, use_cuda ? buffer.out[k] : (records[k]->value - rhs ).squar
                                                                                                                          ^
point_cloud/applications/CMakeFiles/points-join.dir/build.make:484: recipe for target 'point_cloud/applications/CMakeFiles/points-join.dir/points-join.cpp.o' failed
make[3]: *** [point_cloud/applications/CMakeFiles/points-join.dir/points-join.cpp.o] Error 1
CMakeFiles/Makefile2:294: recipe for target 'point_cloud/applications/CMakeFiles/points-join.dir/all' failed
make[2]: *** [point_cloud/applications/CMakeFiles/points-join.dir/all] Error 2
CMakeFiles/Makefile2:306: recipe for target 'point_cloud/applications/CMakeFiles/points-join.dir/rule' failed
make[1]: *** [point_cloud/applications/CMakeFiles/points-join.dir/rule] Error 2
Makefile:234: recipe for target 'points-join' failed
make: *** [points-join] Error 2
`

david-jennings, 2017-09-12:
It looks like this is the guilty commit:

ed39654 - points-join: changes after review

spotiris, 2017-09-12:
FYI this is the fix I made, I haven't ever used points-join so this is tested to the point where it compiles only.
```
diff --git a/point_cloud/applications/points-join.cpp b/point_cloud/applications/points-join.cpp
index 7032615..6fffe09 100644
--- a/point_cloud/applications/points-join.cpp
+++ b/point_cloud/applications/points-join.cpp
@@ -220,7 +220,7 @@ template <> struct traits< Eigen::Vector3d >
         #ifdef SNARK_USE_CUDA
             snark::cuda::buffer buffer;
             void calculate_squared_norms( const Eigen::Vector3d& rhs ) { snark::cuda::squared_norms( rhs, buffer ); }
-            boost::optional< std::pair< Eigen::Vector3d, double > > nearest_to( const point_input& rhs, unsigned int k ) const { return std::make_pair( records[k]->value, use_cuda ? buffer.out[k] : (records[k]->value - rhs ).squaredNorm() ); }
+            boost::optional< std::pair< Eigen::Vector3d, double > > nearest_to( const point_input& rhs, unsigned int k ) const { return std::make_pair( records[k]->value, use_cuda ? buffer.out[k] : (records[k]->value - rhs.value ).squaredNorm() ); }
         #else // SNARK_USE_CUDA
             boost::optional< std::pair< Eigen::Vector3d, double > > nearest_to( const point_input& rhs, unsigned int k ) const
             {

```

vlaskine, 2017-09-12:
Thanks a lot, Steve; it seems to me a recently introduced typo. Could you please review your fix with Matthew Imhoff (@mimhoff) and ask him to push the fix for you. Thanks again!

mimhoff, 2017-09-13:
Thanks @spotiris for spotting this.

Just a note on this code: points-join with normals has not yet been implemented for CUDA builds.

closed 2017-09-13 by mimhoff