acfr/snark #127: tbb/pipeline.h: No such file or directory

josch, 2017-09-24:
Hi,

when building snark I'm getting:

```
[ 11%] Building CXX object point_cloud/applications/CMakeFiles/points-foreground-partitions.dir/points-foreground-partitions.cpp.o
/root/snark/point_cloud/applications/points-foreground-partitions.cpp:44:26: fatal error: tbb/pipeline.h: No such file or directory
 #include <tbb/pipeline.h>
```

But this shouldn't happen as cmake should check for tbb before compilation starts.

josch, 2017-09-24:
Same for pcap.h:

```
[ 75%] Building CXX object sensors/lidars/velodyne/CMakeFiles/snark_velodyne.dir/impl/pcap_reader.cpp.o
In file included from /root/snark/sensors/lidars/velodyne/impl/pcap_reader.cpp:33:0:
/root/snark/sensors/lidars/velodyne/impl/pcap_reader.h:37:18: fatal error: pcap.h: No such file or directory
 #include <pcap.h>
```

The check for pcap.h also should happen in cmake.

vlaskine, 2017-09-26:
@josch : Thank you, a fair comment, checking the installed packages is not implemented (yet) in comma and snark; however it probably is a better practice to implement those at the packaging stage rather than adding these checks to cmake (which may be quite verbose). I have added this item to our internal backlog and therefore am closing this ticket.

closed 2017-09-26 by vlaskine

josch, 2017-09-26:
What should be happen in CMake is a simple `find_package(mylibrary REQUIRED)` call if the dependency comes with a `mylibrary.cmake` file, or if it doesn't using `pkg_check_modules` to use pkg-config. You are already doing the former for your other dependencies.

vlaskine, 2017-09-26:
Thank you, @josch, we understand that. Due to a fairly complex structure of various switches in snark cmake files as well as a panoply of dependencies (not only libraries), it is not one-size-fits-all. One force is the simple uncluttered cmake. The opposing force is that if you have only a couple of applications in different subdirectories using, say tbb, you don't want to make the whole snark dependent on tbb, even if the user switches off all the libraries that make use of it. Also, proper packaging may be a better solution and therefore the right place for it may be not cmake, but, say, ansible scripts. Therefore currently it is not done wrong, but simply is left undone until we have time to properly address it. I said, I have added a corresponding item to our internal backlog.