acfr/snark #128: CMake Error Qt3D_LIB

jeannief, 2018-04-12:
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Qt3D_LIB
    linked by target "view-points" in directory ~/src/snark/graphics/applications
    linked by target "label-points" in directory ~/src/snark/graphics/applications
    linked by target "snark_graphics_qt3d" in directory ~/src/snark/graphics/qt3d

Modified paths above with ~/ .
Ubuntu 16.04 LTS
qt4-default

Help appreciated.

vlaskine, 2018-04-12:
Do you use qt4 (as it seems)? If so, have you installed qt3d? See the installation procedure here:

https://github.com/acfr/snark/wiki/Snark-for-dummies,-on-ubuntu-debian#Qt

Please let us know if you have more questions.

jeannief, 2018-04-12:
Hi,

Thanks. All good now. I had read that Qt3D was installed as part of qt4. My mistake.
Thanks for your help.

vlaskine, 2018-04-13:
Cheers, thanks for reporting it. Qt5.5 and higher has 3D as its part.

closed 2018-04-13 by vlaskine