acfr/snark #131: Cannot build due to libboost library linking

JacobMackay, 2020-04-21:
Trying to build gives this issue and I can't solve it
/usr/bin/ld: cannot find -lBoost::regex

vlaskine, 2020-04-21:
which boost version are you using? are you on ubuntu?

vlaskine, 2020-04-21:
also, do you have comma built and installed?

JacobMackay, 2020-04-21:
Hi Vlad, I've tried boost 1.71.0 and 1.62.0, and yes on ubuntu.
Comma is built and installed. Should it be built with zmq on? ( This can now be installed using sudo apt install libzmq3-dev). This also causes issues with comma build.

vlaskine, 2020-04-21:
i have not tried boost 1.71, but 1.62 should build with no problem
(snark does not depend on the zmq part of comma)
i would recommend:
- check whether you have on ONE boost installed (check both /usr/include/boost and /usr/local/include/boost to make sure
- if two versions of boost are installed, i would recommend to remove them both from include and lib and reinstall boost using apt-get install
- remove everything in your snark repository and run cmake and build from scratch

since you are in ACFR, you could ask someone who already has comma/snark installed for help (i am not with ACFR anymore and don't maintain their version of comma/snark)

david-jennings, 2020-04-21:
Ubuntu which version? I'm curious how you've got versions 1.71 and 1.62 
installed.

Also, try:

$ dpkg -l libboost-regex-dev

Dave.

JacobMackay, 2020-04-21:
I don't have them simultaneously, just tried both. Using 1.71 now.

I fixed it with a dirty hack:
sudo ln -s /usr/lib/x86_64-linux-gnu/libboost_regex.so /usr/lib/x86_64-linux-gnu/libBoost::regex.so

JacobMackay, 2020-04-21:
I think cmake was looking for capitalB Boost...

david-jennings, 2020-04-21:
I'd be a bit surprised if that's normally necessary, that's why I was 
curious about which version of Ubuntu and how boost was installed.

You should be able to just install libboost-all-dev.

JacobMackay, 2020-04-21:
I was confused which to install from all the documentation. But libboost-all-dev installs with no problems, but this just didn't build properly until the dodgy symlink.

david-jennings, 2020-04-21:
I'm still curious, Ubuntu version?

david-jennings, 2020-04-21:
Final question: which application was building when you got the linking error?

david-jennings, 2020-04-22:
There's an unmerged PR to fix this issue at https://github.com/acfr/snark/pull/130/commits/e195dd31f91e9e47e9a9d202b8ed4d6cbae8bc1f.

The patch looks reasonable but I haven't seen the issue which is why I'd like to know how to replicate.

vlaskine, 2020-04-22:
thanks, Dave, i added regex to orthographic/snark (just typed by hand, no merging), although boost regex never has been a problem in orthographic/snark anyway

i cannot comment on anything in acfr/snark